LiquidCGS/FastIMU #2: Does this work with BMX160?

SujithChristopher, 2023-02-24:
Does this library work with BMX160? and give our calibrated Quaternion output?

Because the sensor is not mentioned in the comments. 

Thanks.

LiquidCGS, 2023-03-30:
You might get that sensor to work as a BMI160, however if you do that you won't be able to use the magnetometer on it.

SujithChristopher, 2023-03-31:
Okay, thank you for the reply, 

any plans on supporting, the full sensor in the future?

SujithChristopher, 2023-03-31:
Secondly, in the comments its mentioned.

Move IMU in figure 8 pattern until done!

Do you have any links/documentation what does this figure 8 pattern mean?

This will be helpful, thanks again.

LiquidCGS, 2023-04-01:
I'll add support to it if I can get my hands on one of them. During calibration the sketch will ask you to move the IMU in a figure 8 pattern to calibrate the magnetometer, it quite literally means, to pick up the IMU and move it in the air, drawing an "8" with it while rotating it to calculate the average magnetometer soft iron corrections.

closed 2023-04-09 by LiquidCGS