LiquidCGS/FastIMU #5: MPU6886::calibrateAccelGyro() failure

mhaberler, 2023-04-17:
Hi,
trying [Calibrated_sensor_output.ino](https://github.com/LiquidCGS/FastIMU/blob/main/examples/Calibrated_sensor_output/Calibrated_sensor_output.ino) with an esp32 and an MPU6886

MPU6886::calibrateAccelGyro() fails to read back the FIFO sample count [here](https://github.com/LiquidCGS/FastIMU/blob/main/src/F_MPU6886.cpp#L214) - the readBytes() request is NAK'd, hence fifo_count is zero, resulting in a divide-by-zero [here](https://github.com/LiquidCGS/FastIMU/blob/main/src/F_MPU6886.cpp#L236)

[image]

not sure what to do about it, short from replacing gyro calibration with averaging

thanks!
Michael

ps: using Platformio, esp_wrover_kit 4.1 with JTAG, Salea LA
versions  are:
````
Platform espressif32 @ 6.1.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20007.0 (required: platformio/framework-arduinoespressif32 @ ~3.20007.0)
├── tool-cmake @ 3.16.4 (required: platformio/tool-cmake @ ~3.16.0)
├── tool-esptoolpy @ 1.40500.0 (required: platformio/tool-esptoolpy @ ~1.40500.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.9.0 (required: platformio/tool-ninja @ ^1.7.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)
````

bobobo1618, 2023-11-21:
I'm having the same problem. When I remove the calibration, I get data but they're extremely noisy.

LiquidCGS, 2023-11-29:
I'll see if I can get my hands on an MPU6886 again to properly fix that IMU.

floBik, 2024-07-04:
Please check if my PR #25 solves this Issue.
It works for me, but I'm not sure if the calibration is working properly now.