LiquidCGS/FastIMU #8: An error occurred while compiling

jiahao2333, 2023-07-25:
*Ardunio 2.1.1*

*Fast IMU 1.2.1*

An error occurred while compiling
```
In file included from c:\Users\11984\Documents\Arduino\libraries\FastIMU\src/FastIMU.h:18,
                 from C:\Users\11984\Documents\Arduino\sketch_jul25a\sketch_jul25a.ino:1:
c:\Users\11984\Documents\Arduino\libraries\FastIMU\src/F_QMC5883L.hpp: In member function 'virtual int QMC5883L::setGyroRange(int)':
c:\Users\11984\Documents\Arduino\libraries\FastIMU\src/F_QMC5883L.hpp:45:40: error: no return statement in function returning non-void [-Werror=return-type]
   45 |  int setGyroRange(int range) override {};
      |                                        ^
c:\Users\11984\Documents\Arduino\libraries\FastIMU\src/F_QMC5883L.hpp: In member function 'virtual int QMC5883L::setAccelRange(int)':
c:\Users\11984\Documents\Arduino\libraries\FastIMU\src/F_QMC5883L.hpp:46:41: error: no return statement in function returning non-void [-Werror=return-type]
   46 |  int setAccelRange(int range) override {};
      |                                         ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1
```
The file with compilation error is examples/Calibrated_HadesVR/Calibrated_HadesVR.ino
My current setup is preferences->compiler warnings->none
Is my problem?🤔
Thanks.

LiquidCGS, 2023-07-25:
Ah, Im an idiot, will fix in a minute

closed 2023-07-25 by LiquidCGS

jiahao2333, 2023-07-25:
Cool! Thank