LiquidCGS/FastIMU #10: Does not work with builtin IMU on Nano 33 IoT

Chedski, 2023-08-13:
FastIMU does not work with the builtin IMU of the Nano 33 IoT because the "who am i" value of the LSM6DS3 on the 33 is (or can be?) 0x6C instead of 0x69.

LiquidCGS, 2023-08-13:
That would actually point to that IMU not being an LSM6DS3 and being an LSM6DS0 instead.
https://www.st.com/resource/en/datasheet/lsm6dso.pdf (see 9.11 on page 49)
Seeing as it's register map is pretty similar to the LSM6DS3's you might be able to modify `F_LSM6DS3.hpp`'s line 53 to 0x6C and have it work
I'll look into adding support for it if I can get my hands on one of them.

quicksketch, 2025-01-10:
@LiquidCGS Thank you for the helpful tip. I have a LSM6DSO in a custom board and I confirmed that the `LSM6DS3` driver works well for the accelerometer. The only line that needed changing was this:

```
#define LSM6DS3_WHOAMI_DEFAULT_VALUE	0x69
```

I changed it to this:

```
#define LSM6DS3_WHOAMI_DEFAULT_VALUE	0x6C
```

And then declared that as my IMU class via `LSM6DS3 IMU;` and it worked great with the LSM6DSO accelerometer.

quicksketch, 2025-01-10:
I confirmed this value from the [official Arduino_LSM6DSOX](https://github.com/arduino-libraries/Arduino_LSM6DSOX) library. Check this line in their source code file:

https://github.com/arduino-libraries/Arduino_LSM6DSOX/blob/0586b99357d41ee5c33b5a68e3d9affb9e10920f/src/LSM6DSOX.cpp#L82C48-L82C52

Here's the line copied for convenience:

```
  if (!(readRegister(LSM6DSOX_WHO_AM_I_REG) == 0x6C || readRegister(LSM6DSOX_WHO_AM_I_REG) == 0x69)) {
```

Interestingly they have the conditional that allows either of these addresses. It may be that they interchange the `LSM6DS3` and `LSM6DSO` on their boards and they are hardware-compatible.