LiquidCGS/FastIMU #11: Seems to be an issue with a library manifest and can't be built with Platformio

berkut0, 2023-09-11:
I'm trying to port my source code to Platformio and encountered couple of weird things
1. First of all it can't be found in Platformio library search
2. Platformio "ignores broken library manifest"

[image]

3. For now, I can't build my project directly using the library from github

[image]
<details><summary>screenshots</summary>
Was used

```C
MPU6050 IMU;
```

[image]

[image]

</details>

berkut0, 2023-09-11:
At least one error here:
[image]

LiquidCGS, 2023-09-13:
I don't usually use platformio so I didn't catch this earlier oops
I'll take a closer look at this later today or tomorrow

LiquidCGS, 2023-09-14:
I forgot to make a branch for this fix, it was just that line breaking the manifest. looks like it works now, you'll get some warnings due to me leaving an unfinished IMU type in as it's a WIP but it'll compile just fine

closed 2023-09-14 by LiquidCGS