LiquidCGS/FastIMU #15: MPU-6050 not recognized

SysWhiteDev, 2023-12-31:
I have a MPU-6050 and was trying to make it work with FastIMU, unfortunately i get an error code of -1, that lead me to thinking that my sensor was broken, but when I try to read raw data from it I get a response, also when I scan i2c connections i get a positive one at address 0x68

(Image of output from [Nick Gammon's i2c scanner](https://www.gammon.com.au/forum/?id=10896&reply=6#reply6)) 
[image]

(Image of output from the IMU identifier)
[image]

LiquidCGS, 2024-01-02:
Hey! so basically the way IMUIdentifier (and every single IMU on startup) checks what kind of IMU you have is by scanning the WHOAMI register and comparing it with the expected one from each IMU, say if I scan register 0x75 on an unknown IMU and get 0x68 as a response, I know that 0x68 means it's an MPU6050, if I get 0x70 it's an MPU6500 and so on and so forth, so the reason FastIMU isn't seeing it is because whatever it responded with isn't in it's short database of IMU's

Same happens when you start up an IMU that's already been defined, on initialization it checks that you have the right IMU and when it gets something different than what it expects it stops the initialization. that's why it returns -1.

Some IMU's might say they are one thing while actually being another, there's also a bit of an issue with counterfit IMU's that don't return the value of any known ones... You can try using IMU_GENERIC as an IMU type for these, this type should work with most Invensense IMU's and it doesn't check for IMU type on startup

Also please if you could, try creating a new sketch and run this code and let me know what it returns, maybe it's an IMU Invensense actually makes and could be added to FastIMU as a supported IMU in the future.

```
#include <Wire.h>

#define ADR 0x68
#define REG 0x75

void setup() {
  Serial.begin(9600);
  while (!Serial) ;
  Wire.begin();
}

void loop() {
  Serial.print("Reading address 0x");
  Serial.print(ADR, HEX);
  Serial.print(", Register 0x");
  Serial.println(REG, HEX);

  Serial.print("Response: 0x");
  Serial.println(readByte(ADR, REG), HEX);
  Serial.println("==========================");
  delay(1000);
}

uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data;                            // data will store the register data
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);     // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, (uint8_t) 1, true); // Read one byte from slave register address
  if (Wire.available()) {
    data = Wire.read();                      // Fill Rx buffer with result
  }
  return data;                             // Return data read from slave register
}
```

SysWhiteDev, 2024-01-02:
I ran the script you provided and this is the output:

[image]

SysWhiteDev, 2024-01-02:
My sensor just decided to do some magic smoke and shut down, I'm assuming it completely burnt down and I will not be able to work on it anymore.

closed 2024-01-02 by SysWhiteDev

LiquidCGS, 2024-01-03:
Well that sucks, either way I can't find any reference of a 0x72 response belonging to a non-counterfeit IMU other than an odd google doc saying it belongs to an MPU 9350... which I can't find any info about...
I'll add it to the IMU list as a counterfeit one.