LiquidCGS/FastIMU #21: Divide by Zero during Calibration

dennyfmn, 2024-05-31:
Using FastIMU 1.2.6 on Arduino 2.3.2 with MPU6886 built into M5StickC Plus2. Runs fine if I comment out the Calibration line. If I let it try to calibrate I get:

FastIMU calibration & data example
Keep IMU level.
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d1d0e  PS      : 0x00060c30  A0      : 0x800d18a0  A1      : 0x3ffb21f0  
A2      : 0x00000000  A3      : 0x3ffc1068  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000002  A11     : 0x00000072  A12     : 0x00000002  A13     : 0x3ffb2200  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008700c  LEND    : 0x40087022  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1d0b:0x3ffb21f0 0x400d189d:0x3ffb2230 0x400d5982:0x3ffb2290

ELF file SHA256: 5ee44eb45d02f904

Rebooting...

patrickstreit, 2024-06-17:
I experience the same on a M5Atom Matrix with MPU6886 built in.
Also on Arduino 2.3.2 / FastIMU 1.2.6
Runs fine without startup calibration

> FastIMU calibration
> Keep IMU level.
> Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x400d368b  PS      : 0x00060a30  A0      : 0x800d295e  A1      : 0x3ffb21e0  
> A2      : 0x00000000  A3      : 0x3ffc3844  A4      : 0x00000000  A5      : 0x3ffb2200  
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000072  A13     : 0x00000002  
> A14     : 0x3ffb2200  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000006  
> EXCVADDR: 0x00000000  LBEG    : 0x40089b78  LEND    : 0x40089b8e  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x400d3688:0x3ffb21e0 0x400d295b:0x3ffb2230 0x400d6fe2:0x3ffb2290
> 
> ELF file SHA256: 833ca81c6f4778fc
> 
> Rebooting...

floBik, 2024-07-04:
Hello all,

this Issue should be solved with following PR #25

closed 2024-11-12 by LiquidCGS