LiquidCGS/FastIMU #29: XIAO NRF52840 Sense support?

monokuroumu, 2024-11-02:
I'm trying to build a BLE Headtracker for gaming using XIA BLE Sense - it has an onboard LSM6DS3TR-C.
I try to run IMUIdentifier - and I'm getting "No IMU detected". Is there a possibility to add support for this sensor?
Here's the datasheet for it - https://files.seeedstudio.com/wiki/XIAO-BLE/ST_LSM6DS3TR_Datasheet.pdf - and I can see it uses 0x6A as I2C address for the sensor.

10MINT, 2024-11-20:
Same problem here

quicksketch, 2025-01-12:
In my own use, the `LSM6DSL` driver works perfectly with a `LSM6DS3TR-C`. I know you'd think that it would work with the `LSM6DS3` driver since it has a more similar model number, but apparently it's the `LSM6DSL` and `LSM6DS3TR-C` that are essentially interchangeable.

10MINT, 2025-01-23:
That doesn't work for me. Setting
```cpp
#define IMU_ADDRESS 0x6A
LSM6DSL IMU;
```
in the examples gives the following console output: `Error initializing IMU: -1`

Maybe something weird with the addressing on the nrf52