LiquidCGS/FastIMU #31: Plans to add LSM6DS3TR-C ?

masynmachien, 2024-11-11:
Thank you very much for making and sharing this library.

I have been using your FastIMU library succesfully to move our project https://github.com/FedericoBusero/Wifi-Hovercraft-Browser from the MPU-6050 (no longer manufactured) to newer cheap IMU's.
I had great results with the LSM6DS3TR, but that one is also no longer manufactured.
I switched to the LSM6DS3TR-C, but it does not initialise.
I do get it to work with the Adafruit_LSM6DS3TRC library, using I2C (my preference)
Good to know probably, is that the Adafruit_LSM6DS33 library can use any LSM6DS/ISM330 family chip, EXEPT the LSM6DS3TR-C.

Thanks and best regards,

masynmachien

FedericoBusero, 2024-11-11:
That would be great! In fact it is perfectly compatible with LSM6DS3.

As a worlaround, simply remove following lines and it works.

```
	if (!(readByte(IMUAddress, LSM6DS3_WHO_AM_I) == LSM6DS3_WHOAMI_DEFAULT_VALUE)) {
		return -1;
	}
```

LiquidCGS, 2024-11-12:
Yep, LSM IMU family is big and *for the most part* shares registers so that temporary workaround should be fine for now.
I'll add proper support for it in the following days.

closed 2024-11-12 by LiquidCGS