LiquidCGS/FastIMU #37: Compilation issue

linhz0hz, 2024-12-14:
I have compilation issue compiling the example script on Arduino 2.3.4 and EPS32-C6

`
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMX055.cpp:3:5: error: default argument given for parameter 2 of 'int BMX055::init(calData, uint8_t)' [-fpermissive]
    3 | int BMX055::init(calData cal, uint8_t address = BMX055_DEFAULT_ADDRESS)
      |     ^~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMX055.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMX055.hpp:69:13: note: previous specification in 'virtual int BMX055::init(calData, uint8_t)' here
   69 |         int init(calData cal, uint8_t address = BMX055_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_HMC5883L.cpp:3:5: error: default argument given for parameter 2 of 'int HMC5883L::init(calData, uint8_t)' [-fpermissive]
    3 | int HMC5883L::init(calData cal, uint8_t address = HMC5883L_DEFAULT_ADDRESS)
      |     ^~~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_HMC5883L.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_HMC5883L.hpp:36:13: note: previous specification in 'virtual int HMC5883L::init(calData, uint8_t)' here
   36 |         int init(calData cal, uint8_t address = HMC5883L_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI055.cpp:3:5: error: default argument given for parameter 2 of 'int BMI055::init(calData, uint8_t)' [-fpermissive]
    3 | int BMI055::init(calData cal, uint8_t address = BMI055_DEFAULT_ADDRESS)
      |     ^~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI055.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI055.hpp:48:13: note: previous specification in 'virtual int BMI055::init(calData, uint8_t)' here
   48 |         int init(calData cal, uint8_t address = BMI055_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_IMU_Generic.cpp:5:5: error: default argument given for parameter 2 of 'int IMU_Generic::init(calData, uint8_t)' [-fpermissive]
    5 | int IMU_Generic::init(calData cal, uint8_t address = IMU_Generic_DEFAULT_ADDRESS)
      |     ^~~~~~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_IMU_Generic.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_IMU_Generic.hpp:169:13: note: previous specification in 'virtual int IMU_Generic::init(calData, uint8_t)' here
  169 |         int init(calData cal, uint8_t address = IMU_Generic_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20689.cpp:5:5: error: default argument given for parameter 2 of 'int ICM20689::init(calData, uint8_t)' [-fpermissive]
    5 | int ICM20689::init(calData cal, uint8_t address = ICM20689_DEFAULT_ADDRESS)
      |     ^~~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20689.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20689.hpp:91:13: note: previous specification in 'virtual int ICM20689::init(calData, uint8_t)' here
   91 |         int init(calData cal, uint8_t address = ICM20689_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DSL.cpp:5:5: error: default argument given for parameter 2 of 'int LSM6DSL::init(calData, uint8_t)' [-fpermissive]
    5 | int LSM6DSL::init(calData cal, uint8_t address = LSM6DSL_DEFAULT_ADDRESS)
      |     ^~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DSL.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DSL.hpp:64:13: note: previous specification in 'virtual int LSM6DSL::init(calData, uint8_t)' here
   64 |         int init(calData cal, uint8_t address = LSM6DSL_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK09918.cpp:4:5: error: default argument given for parameter 2 of 'int AK09918::init(calData, uint8_t)' [-fpermissive]
    4 | int AK09918::init(calData cal, uint8_t address = AK09918_DEFAULT_ADDRESS)
      |     ^~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK09918.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK09918.hpp:66:13: note: previous specification in 'virtual int AK09918::init(calData, uint8_t)' here
   66 |         int init(calData cal, uint8_t address = AK09918_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI160.cpp:5:5: error: default argument given for parameter 2 of 'int BMI160::init(calData, uint8_t)' [-fpermissive]
    5 | int BMI160::init(calData cal, uint8_t address = BMI160_DEFAULT_ADDRESS)
      |     ^~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI160.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_BMI160.hpp:111:13: note: previous specification in 'virtual int BMI160::init(calData, uint8_t)' here
  111 |         int init(calData cal, uint8_t address = BMI160_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8975.cpp:4:5: error: default argument given for parameter 2 of 'int AK8975::init(calData, uint8_t)' [-fpermissive]
    4 | int AK8975::init(calData cal, uint8_t address = AK8975_DEFAULT_ADDRESS)
      |     ^~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8975.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8975.hpp:40:13: note: previous specification in 'virtual int AK8975::init(calData, uint8_t)' here
   40 |         int init(calData cal, uint8_t address = AK8975_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DS3.cpp:5:5: error: default argument given for parameter 2 of 'int LSM6DS3::init(calData, uint8_t)' [-fpermissive]
    5 | int LSM6DS3::init(calData cal, uint8_t address = LSM6DS3_DEFAULT_ADDRESS)
      |     ^~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DS3.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_LSM6DS3.hpp:64:13: note: previous specification in 'virtual int LSM6DS3::init(calData, uint8_t)' here
   64 |         int init(calData cal, uint8_t address = LSM6DS3_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20690.cpp:5:5: error: default argument given for parameter 2 of 'int ICM20690::init(calData, uint8_t)' [-fpermissive]
    5 | int ICM20690::init(calData cal, uint8_t address = ICM20690_DEFAULT_ADDRESS)
      |     ^~~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20690.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_ICM20690.hpp:92:13: note: previous specification in 'virtual int ICM20690::init(calData, uint8_t)' here
   92 |         int init(calData cal, uint8_t address = ICM20690_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8963.cpp:4:5: error: default argument given for parameter 2 of 'int AK8963::init(calData, uint8_t)' [-fpermissive]
    4 | int AK8963::init(calData cal, uint8_t address = AK8963_DEFAULT_ADDRESS)
      |     ^~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8963.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_AK8963.hpp:41:13: note: previous specification in 'virtual int AK8963::init(calData, uint8_t)' here
   41 |         int init(calData cal, uint8_t address = AK8963_DEFAULT_ADDRESS) override;
      |             ^~~~
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_MPU6050.cpp:5:5: error: default argument given for parameter 2 of 'int MPU6050::init(calData, uint8_t)' [-fpermissive]
    5 | int MPU6050::init(calData cal, uint8_t address = MPU6050_DEFAULT_ADDRESS)
      |     ^~~~~~~
In file included from c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_MPU6050.cpp:1:
c:\Users\linhz\Documents\Arduino\libraries\FastIMU\src\F_MPU6050.hpp:78:13: note: previous specification in 'virtual int MPU6050::init(calData, uint8_t)' here
   78 |         int init(calData cal, uint8_t address = MPU6050_DEFAULT_ADDRESS) override;
      |             ^~~~

exit status 1

Compilation error: exit status 1
`

LiquidCGS, 2024-12-14:
Will look into this in a few minutes.

LiquidCGS, 2024-12-14:
Fixed, 1.2.8 of FastIMU should show up in the library manager in a few hours.

closed 2024-12-14 by LiquidCGS